openMLA/elegoo-mars-4-dlp-controller #4: Soldering 5V port on PCB

Hackerman2016, 2024-06-09:
So I have built the PCB according to the instructions, and I was wondering whether or not to solder the 5V port on the PCB, and after the PCB is complete, what is ideally the next step? I have to install the packages as specified on the pi and run the main.py, and it should work?

NemoAndrea, 2024-06-09:
The 5V pin on the PCB (assuming we are talking about the same pin), can be left out; it is just there in case you want to power something with the 5V from the pi/dcdc converter. 

If you are taking about the solder bridge: it should be soldered if you have a the  dc-dc converter (CUI P78E)  soldered in place; if you don't have the CUI, then it doesn't really matter if you close the solder bridge. 

> [!WARNING]
> Whatever you do, do not power the raspberry pi through USB and the CUI converter at the same time. It is **either** power through the USB connector of the pi (leave the 5V solder bridge empty, and leave out the CUI) **or** power through the 12V barrel jack with the CUI to provide the 5V power for the Pi.

If you have assembled the board and checked that the pi still works (I suggest you do this before connecting the projector). then you hook up the projector, follow the software packages instructions and then indeed try main.py.

If that all works you can start writing your own python code for your project  using the included code :)

Hackerman2016, 2024-06-10:
I have the dc-dc converter in place.
[image]
This is the 5V port I am talking about.

NemoAndrea, 2024-06-10:
Ah yes, that one you should bridge if you have the dc-dc converter installed. This connects the 12V to the input of the converter. 

To reiterate (to prevent damage), do not connect USB power to the pi if you also connect the 12V at the same time.

Hackerman2016, 2024-06-11:
Ah, okay. Great. Thanks a lot.

closed 2024-06-11 by Hackerman2016