openMLA/elegoo-mars-4-dlp-controller #5: Regarding running the python code

Hackerman2016, 2024-06-11:
[image]

I have connected all the components and have managed to power on the printer after connecting the pcb and pi but as soon as I run the code I get this error. Any help with this will be appreciated. Also I have trouble in connecting the pi to the internet. Doing configuring the rasbian os 64 bit on the pi zero 2W I did the ssh settings but still somehow when I login on putty with the hostname I get an error saying hostname doesn't exist.

Hackerman2016, 2024-06-11:
Also when I am trying to increase the SPI buffer size I followed the steps mentioned in the document but I am unable to do it. I am getting this error.
[image]

NemoAndrea, 2024-06-25:
Hmm for the first point, could it be that you are running the python code on your local machine and not on the raspberry pi? If not then you need to install RPi.GPIO but I think that should definitely come with the default OS.

For the 2nd image, why is there `firmware` in the path?  It should just be `/boot/cmdline.txt`; that might fix it

Hackerman2016, 2024-06-26:
For the first point you were right I was trying to run it on local machine. Running it on Pi solved the issue and yeah about making changes in the /boot/cmdline.txt file, for me after running that command I get an error saying that the file has been shifted to /boot/firmware/cmdline.txt. That's why increasing the buffer size and stuff I have all done it in that file.

NemoAndrea, 2024-06-26:
Hmm, strange, I am not aware of this change. I guess there must be a file `cmdline.txt` somewhere on the system. Maybe you can post raspberry pi os version here, so that other people can figure out if they may be running a version with issues too.

Hackerman2016, 2024-06-27:
That is a difference I am not working with the OS lite and Pi Zero. I am working with the Raspberry Pi 3b+ and the standard Raspberry OS 64-bit.

NemoAndrea, 2024-08-06:
Okay, I will close this issue, as I think it would be better to make a more focussed new issue if you cannot figure it out. Nice to see you being able to get the project working (at least in parts!). 

If you run into issues where the documentation is wrong or even just unclear, let me know because that is super valuable for the project :)

closed 2024-08-06 by NemoAndrea