openMLA/elegoo-mars-4-dlp-controller #6: Error while reading i2c port 1

Hackerman2016, 2024-06-24:
Hello,

I am trying to run the python script and I am getting an error where the i2c 1 port is undetectable.  I have tried adding these lines in the boot file 'dtparam=i2c_arm=on
dtparam=i2c1=on'   

Still I get the same error. Here is the error that I get:

  File "/home/pi/Desktop/UV_projector/controller.py", line 61, in __init__
    i2c_bus.read_byte(self.addr)  # check if i2c is already active. Then we are in [2]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/main.py", line 24, in <module>
    DMD = DLPC1438(i2c, spi)
          ^^^^^^^^^^^^^^^^^^
  File "/home/pi/Desktop/UV_projector/controller.py", line 85, in __init__
    while i2c_bus.read_byte(self.addr) == 0:  
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error

NemoAndrea, 2024-06-25:
hmm, what does `i2cdetect -y 1` give?

Hackerman2016, 2024-06-26:
It gives a table where there are no addresses assigned. So it is just a table with dashes.

NemoAndrea, 2024-06-26:
Okay, so that means that the I2C hardware on the Pi is working.  I don't really know what is causing the issue; it certainly seems like the DLPC1438 is not starting properly. Do you see some LEDs lit up on the mainboard? Does it work if you run the code a few times (Sometimes it needs one run to get going; something not quite right about the power up cycle in the code)

Hackerman2016, 2024-06-28:
[image]

Do i need to solder these adapters too on the pcb or are they optional? What are their functions?

NemoAndrea, 2024-07-17:
> ![image0 (2)](https://private-user-images.githubusercontent.com/170014411/344107721-8f8bc00e-e9fa-4d8c-bf7a-1d914d15774f.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyMTc4NjcsIm5iZiI6MTcyMTIxNzU2NywicGF0aCI6Ii8xNzAwMTQ0MTEvMzQ0MTA3NzIxLThmOGJjMDBlLWU5ZmEtNGQ4Yy1iZjdhLTFkOTE0ZDE1Nzc0Zi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTdUMTE1OTI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDZkYTRiOWJhNDcyMjA0ODJmY2NmNzczMTM1OTYzNWYwOWJjNTlkZTQ0OTRiMTg1NDRlOGFiZGY4ZGU2NDRmOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KolqDCWs3LSlLgdFGFEeP9A25HvDo95ajfdwG2wfoAs)
> 
> Do i need to solder these adapters too on the pcb or are they optional? What are their functions?
> 
> Also does any pin or cable go into the adapter labelled i2c, cause I can't seem to find any that go in that.

Those circled solder points are indeed optional :). They can be used if you find you need those pins for something else (e.g. you want to connect to another SPI peripheral or need some 3.3V power for something).

The I2C connector on the right side in your image is also there just as an optional extra; if you want to connect one of the many i2c breakout boards that use a StemmaQT/QWIIC connector. But it is not needed for operation of the DMD unit.

NemoAndrea, 2024-07-17:
> I did running the code 3-4 times, still same error. That is what I am wondering, that i2cdetect -y 2 shows address and -y 1 doesn't. Does that mean one of the ports is working and one isn't? (1 of the 2 picoclasps that we connect on the pcb). I have tried running test scripts for the 2nd port and they work correctly so I think there has to be maybe an issue on the hardware soldering side or in the connections on the pcb.
> 
> Also yeah LEDs on the DLPC1438 do light up for a while and then after I get the error they turn off.

Hmm this is interesting, I will check if maybe I use i2c bus 2 and forgot to add that to documentation.

Okay, the LED on the DLPC1438 board turning off means that at least one of the hardware connections from the picoclasp is working OK).

NemoAndrea, 2024-07-18:
> I did running the code 3-4 times, still same error. That is what I am wondering, that i2cdetect -y 2 shows address and -y 1 doesn't. Does that mean one of the ports is working and one isn't? (1 of [...]

From what I gather, /dev/i2c/2 is the i2c data for the HDMI connector, and is therefore not showing the board data. /dev/i2c/1 (what you see by `i2cdetect -y 1`) should be i2c bus that you need. That one is also the one the code uses.

So it seems that for whatever reason the elegoo board is not showing up on the i2c bus. That could be for various reasons: the i2c connection having issues, the elegoo board not starting due to faulty connections, the elegoo board not starting because it is damaged etc.  It is a bit surprising since the board is responding to the code - so maybe that points to the i2c connection soldering being off?

Hard to say without really probing the board itself I think. You could doublecheck the solder connections and see if there is anything wrong.

Hackerman2016, 2024-07-31:
Yes indeed, there was a problem with one of pins when I soldered them, fixing that resolved the issue, thanks.

closed 2024-07-31 by NemoAndrea