openMLA/elegoo-mars-4-dlp-controller #7: Programming the Stepper Motor for the Printer

Hackerman2016, 2024-08-13:
I am thinking of also controlling the stepper motor to control the Z stage of the printer but I don't think I can just connect the pins to the custom pcb and modify the python script to run it. Would connecting the stepper motor pins to an Arduino and then somehow configuring that arduino also with the same Raspberry Pi be a solution?

NemoAndrea, 2024-08-14:
The board should have a few spare GPIOs that you can solder headers to. Those you can then use to output step/dir signals over. You would need a simple stepper driver board that takes that output and manages the power delivery and phases for the stepper; but these kinds of boards are very cheap and common. I guess you would need to check a bit that the driver can manage the particular motor used in the elegoo unit.

Hackerman2016, 2024-08-15:
[image]

Do you mean connecting the cables here on this header? I tried directly connecting it on the pcb but I think I will have to find a driver board.

NemoAndrea, 2024-08-15:
Yes, those pins are connected to some GPIO pins (check the PCB schematic to find out which ones). These can provide the directional signal for the stepper, but definitely not the power required to actually actuate the stepper motor. For that indeed you need a little driver board.  You may need to get a bit creative with soldering if the stepper needs 12V (as I didn't put a nice header for the 12V output I think...