openMLA/elegoo-mars-4-dlp-controller #9: Mount 2 upwards pointing cameras connected to raspi5 to align PCB with registration holes?

dirkvdb2, 2025-03-26:
I was brainstorming about another aspect of using specifically the Mars 4 DLP: to find a solution for hole/bottom layer / top layer alignment in dual layer PCB production - then I found your project. Very interesting
My idea was to connect two upward facing raspi camera 3 connected to a raspberry pi 5 to align the drilled blanc pcb with the projected image for the top and botton layer. The mas 4 DLP is particularly interting for this because of the absence of a LCD so an unobstructed view from below the glass plate. With a simple python program, the 2 camera images could be displayed on a single screen and thus allow to align the drilled PCB with the projeced image. The trouble is that the mars 4 DLP is no longer in production and that other DLP printer are too expensive. But I have one and I have a source to buy stock for around 300€.
The combination of your project with tools to aloign would be a killer project.

NemoAndrea, 2025-04-02:
Sounds like an interesting idea where the key challenge would be to map the camera alignment with the DLP exposure plane. Might be relatively doable with a transparent/translucent reference sample to align the camera for backside (side not facing the DLP projector).

If you want to maximise resolution, you could use a hot/coldmirror to keep the mainside camera and the UV light path co-axial. But might need modifications to the original projector so probably not super attractive.

Let me know if there is something particular that you would need my help with; I am really not sure if I can find time to help; but hopefully I will at least be able to assist with matters related to the projector.

NemoAndrea, 2026-09-11:
Did you end up trying this in the end? I'd love to hear about it and maybe it can help someone else too. For now I will close the issue!

closed 2026-09-11 by NemoAndrea